pvtom/rscp2mqtt #89: [Feature Request] PVI idle mode

cwihne, 2024-09-30:
Autumn is here, autarky level gets impacted. S10E has a feature enabled putting PVI into sleep mode when battery is empty and no PV input to expect, e.g. in early morning hours before sundawn. I was trying to find out about PVI idle status, didn't find such in data transmitted to MQTT so far. Yet, I see RSCP tags `TAG_PVI_DEVICE_STATE`, `TAG_PVI_SYSTEM_MODE` or `TAG_PVI_POWER_MODE` which sound promising.
I am monitoring those now by new raw queries. Maybe you have an idea which tag shows idle state of PVI and get it transmitted to MQTT?

pvtom, 2024-10-01:
Hi, I found the following tags in RscpTags.h that may be requesting the required state: TAG_PVI_REQ_STATE, TAG_PVI_REQ_SYSTEM_MODE.
I hope you will find the right one and can let us know in this issue!
Best regards Thomas

cwihne, 2024-10-14:
Here an intermediate result, still under surveillance: 
`TAG_PVI_REQ_POWER_MODE` seems to represent PVI's power state ON (1) or OFF (0)
`TAG_PVI_REQ_DEVICE_STATE` creates sub-topics, most importantly `TAG_PVI_DEVICE_CONNECTED` (device available for service) and `TAG_PVI_DEVICE_WORKING` (device actively working).

Have to query specifically:
ADD_NEW_REQUEST=TAG_PVI_REQ_DATA:TAG_PVI_INDEX:0-0
ADD_NEW_REQUEST=TAG_PVI_REQ_DATA:TAG_PVI_REQ_POWER_MODE-1
ADD_NEW_REQUEST=TAG_PVI_REQ_DATA:TAG_PVI_REQ_DEVICE_STATE-1

Overall E3DC system's own net power consumption for system operation is significantly influenced by PVI's activity.
I continue investigating the flags and consumption.
Btw. the RAW requests are not updated on forced update, they need a service restart to be re-sent.

closed 2025-03-11 by pvtom