pvtom/rscp2mqtt #90: Cycle Time is not the Intervall given in cfg it is 2s more for me.

RobertHerter, 2024-10-22:
Hello,

i use rscp2mqtt for a long time now and started to use PowerManagement ("e3dc/set/power_mode":) to load/idle the Battery with an dynamic tarif.

To load the Battery for an Hour on low charges, i use charge:12000:1800 on an interval set on 2 seconds. i found that it loads nearly  double the time (2h) than one hour.

Looking at the code i found, that you sleep in wsleep for the interval given. But the runtime of one cycle for me is nearly 1.6 seconds so one cycle  is not 2 seconds but 3.6.

as a quick an dirty hack i changed wsleep to 
`void wsleep(double seconds) {

    while(seconds >= 0.0)
    {
        mtx.lock();
        if (mqttRcvd || !go) {
            mtx.unlock();
            return;
        }
        mtx.unlock();
        if(seconds > 1.0) {
          sleep(1);
        }
        else {
          useconds_t usec = (useconds_t)(seconds * 1e6);
          if(usec > 0)
            usleep(usec);
        }
        seconds = seconds - 1.0;
    }
    return;
}`

and measure the runtime within mainLoop at the start of the loop-body with
`    struct timeval start, end;

    while (go && !bStopExecution) {
        gettimeofday(&start, NULL);
`
and at the end with:
`          // Record end time                                                                                                                                                                                                                                    
          gettimeofday(&end, NULL);
          // Calculate the time difference in microseconds                                                                                                                                                                                                      
          long seconds = end.tv_sec - start.tv_sec;
          long microseconds = end.tv_usec - start.tv_usec;
          double elapsed = seconds + microseconds * 1e-6;
          //Calculate remaining wait                                                                                                                                                                                                                            
          elapsed = (double) cfg.interval - elapsed;
          if(elapsed < 0.0)
            elapsed = 0.0;
          wsleep(elapsed);
`

pvtom, 2024-10-22:
Hello Robert,
I can reproduce the problem.
I have adopted your solution :-) and also used the elapsed time instead of counting down the cycles in the power mode functionality. This should result in a more accurate runtime for charging etc.
The solution will come with release v3.31, probably next weekend.
Best regards
Thomas

pvtom, 2024-10-26:
fixed by v3.31

pvtom, 2024-10-26:
Note: The value for the interval should possibly be increased as the frequency of accesses to the home power station increases with the new implementation.

closed 2025-03-11 by pvtom