pvtom/rscp2mqtt #92: rscp2mqtt no longer polled at 30 second intervals but at 3 second intervals

jomifa, 2024-12-27:
Hello,
I discovered today that the poll sequence has been reduced from 30 to 3 seconds. No change was made to the .config file.

Extract from the .config:

#-System
#--cycle interval in seconds (max 300 seconds)
INTERVAL=30

PS
RSCP2MQTT runs as version 3.23

Best regards,
Joachim

pvtom, 2024-12-27:
Hello Joachim,
rscp2mqtt is implemented so that the wait time (set by the INTERVAL parameter) is interrupted when an MQTT command (a set command, e.g. e3dc/set/wallbox/xxx) is received so that it is processed immediately.
Are you using another tool that sends control commands to rscp2mqtt every 3 seconds?

Best regards,
Thomas

jomifa, 2024-12-27:
Hello Thomas,
no, I am not aware that I would send other E3DC commands via MQTT. I followed up on your hint and found a place in my flow that had written the status change via MQTT back into the MQTT (set). This then led to a loop condition. 

Thank you for your support.

Greetings
Joachim

closed 2025-02-02 by pvtom