pvtom/rscp2mqtt #93: Battery energy charged / discharged values

universal-dilettant, 2025-01-03:
Hallo Thomas, gutes neues Jahr!

I'm struggling to identify the right values of the total amount of charged and discharged energy to/from the battery.  Any hint where to find it? I'd have expected 2 battery-related energy counters somewhere, most likely under e3dc/pvi/energy_all.

BTW: It's not very clear, additionally, if some power and energy values are DC or AC. One can only guess. Maybe you want to consider a non-breaking change by adding additional mqtt topics with AC and DC in the names? 

Danke Josef

pvtom, 2025-01-03:
Hallo Josef,
all topics are listed in the TOPICS.md file. Here are the battery energy topics:

| Device / Tag | MQTT Topic | Values / [Unit] |
| --- | --- | --- |
| Battery Energy Charge (today) | e3dc/battery/energy/charge | [kWh] |
| Battery Energy Discharge (today) | e3dc/battery/energy/discharge | [kWh] |
| Historical Data Battery Energy Charge ** | e3dc/day/<year#>/<month#>/<day#>/battery/energy/charge | [kWh] |
| Historical Data Energy Discharge ** | e3dc/day/<year#>/<month#>/<day#>/battery/energy/discharge | [kWh] |
| Month Battery Energy Charge | e3dc/month/battery/energy/charge | [kWh] |
| Month Energy Discharge | e3dc/month/battery/energy/discharge | [kWh] |
| Week Battery Energy Charge | e3dc/week/battery/energy/charge | [kWh] |
| Week Energy Discharge | e3dc/week/battery/energy/discharge | [kWh] |
| Year Battery Energy Charge | e3dc/year/battery/energy/charge | [kWh] |
| Year Energy Discharge | e3dc/year/battery/energy/discharge | [kWh] |
| Year History Battery Energy Charge | e3dc/history/<year#>/battery/energy/charge | [kWh] |
| Year History Energy Discharge | e3dc/history/<year#>/battery/energy/discharge | [kWh] |
| Yesterday Battery Energy Charge | e3dc/yesterday/battery/energy/charge | [kWh] |
| Yesterday Energy Discharge | e3dc/yesterday/battery/energy/discharge | [kWh] |

Regarding AC/DC:
I don't know exactly whether E3/DC determines the power on the DC or AC side. From my point of view topics with “pvi” and “string” are DC values. Everything with L1, L2, L3 are AC values.

Beste Grüße und ebenfalls ein gute Neues Jahr!
Thomas

pvtom, 2025-01-04:
Hi Josef,
please set HST_REQUESTS=true
Gruß Thomas

universal-dilettant, 2025-01-04:
Hi Thomas,
Looks like the _e3dc/battery/energy topic_ has now appeared. Will obeserve the values; at a 1st glance it looks like these are daily values only - would have expected totals like the other _energy_ related values...
Josef

universal-dilettant, 2025-01-14:
Hallo Thomas,
I have been collecting these values for the last 10 days or so. I'm wondering about the physical unit of the battery's energy values? If I'd assume kWh per day, the figures don't match, when comparing with the S10 portal values. You've got an idea about that? My S10 has a capacity of 12kWh, provided by 4 battery modules.
Danke dir.

closed 2025-01-15 by universal-dilettant