pvtom/rscp2mqtt #94: Preisgrenze für das Laden der Hauskraftwerksbatterie setzen

Dirk070, 2025-01-23:
Hallo Thomas,

besteht die Möglichkeit, in den Smart-Funktionen -> SmartCharge -> Preisbasiertes Laden der Hauskraftwerksbatterie die "Preisgrenze für das Laden der Hauskraftwerksbatterie" zu setzen?
Hintergrund: um optimal den geringesten Preis nutzen zu können, muss man aktuell täglich diesen Wert manuell ändern. Liest man die Werte z.B. von Tibber aus, könnte man diese automatisiert anpassen.

Danke voran und schöne Grüße
Dirk

pvtom, 2025-01-23:
Hallo Dirk,
mir ist derzeit kein RSCP-Tag bekannt, mit dem man den Preis über die Schnittstelle setzen kann. Man müsste mal bei E3/DC anfragen, ob es da etwas gibt.
Man könnte natürlich auch selbst eine Logik aufsetzen, die abhängig vom Strompreis das manuelle Laden startet, dafür gibt es ja Funktionen im rscp2mqtt. 
E3/DC hat andererseits eine neue Software angekündigt, die das Handling mit dyn. Stromtarifen optimieren soll. Evtl. löst man das beschriebene Problem damit direkt systemintern. Dann braucht man keine Steuerung von extern.
Gruß Thomas

Dirk070, 2025-01-26:
Hallo Thomas,

Plan B hat mit "e3dc/wallbox/suspended" funktioniert, TOP.
Eine Frage, bei mir kommt kein reading zum "e3dc/wallbox/soc" an.
Wenn ich es richtig verstehe, könnte der SOC des E-Autos hier geliefert werden, das wäre super.
Funktioniert das reading so bei Dir? Wie immer, wenn ich Infos liefern kann, dann gerne.

Schöne Grüße
Dirk

pvtom, 2025-01-26:
Hallo Dirk,
ich habe keine E3/DC-Wallbox, daher kann ich dazu nichts sagen.
Die Frage wäre, kann die E3/DC-Wallbox mit dem Auto kommunizieren und den SOC abfragen?
Das geht evtl. nicht mit jedem Fahrzeug.
Gruß Thomas

Dirk070, 2025-01-27:
Hallo Thomas,

Danke für die Rückmeldung.
Nach Recherche wird für DC-Laden ein anderes Protokoll verwendet, als beim AC. In der Konsequenz kommt der SOC nur beim DC-Laden. Dann gibt es das Feld bei E3/DC vielleicht nur mit (hoffnungsvollem) Blick in die Zukunft.

Danke Dir und schöne Grüße 
Dirk

closed 2025-07-07 by pvtom