pvtom/rscp2mqtt #98: SmartGrid question

jomifa, 2025-02-21:
hello,
is there an MQTT topic which shows the SmartGrid mode?

pvtom, 2025-03-03:
Hello,
do you mean SG Ready? With release v3.34 I have made some adjustments.
Please install v3.34 and insert the following lines into the .config file, after starting the program you will receive output for RG Ready. Please check if it fits.

# SG Ready
ADD_NEW_REQUEST_AT_START=0:TAG_SGR_REQ_HW_PROVIDER_LIST-0
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_INDEX:_:1:0:sg_ready/1/provider/index
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_NAME:_:1:0:sg_ready/1/provider/name
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_INDEX:_:1:0:sg_ready/2/provider/index
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_NAME:_:1:0:sg_ready/2/provider/name
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_INDEX:_:1:0:sg_ready/3/provider/index
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_NAME:_:1:0:sg_ready/3/provider/name
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_INDEX:_:1:0:sg_ready/4/provider/index
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_NAME:_:1:0:sg_ready/4/provider/name
ADD_NEW_REQUEST=0:TAG_SGR_REQ_READY_TO_USE-0
ADD_NEW_TOPIC=0:TAG_SGR_READY_TO_USE:W:1:0:sg_ready/ready_to_use
ADD_NEW_REQUEST=0:TAG_SGR_REQ_COOLDOWN_END-0
ADD_NEW_TOPIC=0:TAG_SGR_COOLDOWN_END:_:1:0:sg_ready/cooldown_end
ADD_NEW_REQUEST=0:TAG_SGR_REQ_USED_POWER-0
ADD_NEW_TOPIC=0:TAG_SGR_USED_POWER:W:1:0:sg_ready/used_power
ADD_NEW_REQUEST=0:TAG_SGR_REQ_GLOBAL_OFF-0
ADD_NEW_TOPIC=0:TAG_SGR_GLOBAL_OFF:_:1:0:sg_ready/global_off
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_INDEX:0-0
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_REQ_STATE-1
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_INDEX:_:1:0:sg_ready/1/index
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_STATE:_:1:0:sg_ready/1/status
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_AKTIV:_:1:0:sg_ready/1/active
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_INDEX:1-2
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_REQ_STATE-3
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_INDEX:_:1:0:sg_ready/2/index
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_STATE:_:1:0:sg_ready/2/status
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_AKTIV:_:1:0:sg_ready/2/active
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_INDEX:2-4
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_REQ_STATE-5
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_INDEX:_:1:0:sg_ready/3/index
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_STATE:_:1:0:sg_ready/3/status
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_AKTIV:_:1:0:sg_ready/3/active
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_INDEX:3-6
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_REQ_STATE-7
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_INDEX:_:1:0:sg_ready/4/index
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_STATE:_:1:0:sg_ready/4/status
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_AKTIV:_:1:0:sg_ready/4/active

Best regards
Thomas

jomifa, 2025-03-03:
Hello Thomas,
when I use the new config parameters for SG Ready I get the following error messages at startup:

rscp2mqtt [3.34]
E3DC system >192.168.178.31:5033< user: >joachim.fach@jmf.de<
MQTT broker >Pi5JMF:1883< qos = >0< retain = >✗< tls >✗< client id >✗< prefix >e3dc<
Requesting PVI ✓ | PM (0) | DCB ✓ (2 battery strings) | Wallbox ✓ (0) | Interval 30 | Autorefresh ✗ | Raw data ✗ | Logging ON
[2025-03-03 16:52:34] pid=92159 ppid=87916 RscpMqttMain.cpp(2228) Error: Container/Tag >TAG_SGR_READY_TO_USE< received >UNKNOWN_TAG< [1]
[2025-03-03 16:52:34] pid=92159 ppid=87916 RscpMqttMain.cpp(2368) Not supported by the device: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_COS_PHI< received >NOT_AVAILABLE< [1]
[2025-03-03 16:52:34] pid=92159 ppid=87916 RscpMqttMain.cpp(2368) Not supported by the device: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_VOLTAGE_MONITORING< received >NOT_AVAILABLE< [1]
[2025-03-03 16:52:34] pid=92159 ppid=87916 RscpMqttMain.cpp(2368) Not supported by the device: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_FREQUENCY_UNDER_OVER< received >NOT_AVAILABLE< [1]
[2025-03-03 16:52:36] pid=92159 ppid=87916 RscpMqttMain.cpp(2228) Error: Container/Tag >TAG_SGR_READY_TO_USE< received >UNKNOWN_TAG< [1]
[2025-03-03 16:52:36] pid=92159 ppid=87916 RscpMqttMain.cpp(2228) Error: Container/Tag >TAG_DB_HISTORY_DATA_YEAR< received >UNKNOWN< [1]
[2025-03-03 16:52:36] pid=92159 ppid=87916 RscpMqttMain.cpp(2231) History start year has been corrected >2023<
[2025-03-03 16:54:30] pid=92159 ppid=87916 RscpMqttMain.cpp(2228) Error: Container/Tag >TAG_SGR_READY_TO_USE< received >UNKNOWN_TAG< [5]

I don't know whether these are partly errors from version 3.34 or whether I already got the errors from version 3.32. I have nothing to do with containers.

Regards,
Joachim

pvtom, 2025-03-04:
Hello Joachim,
okay, I can confirm TAG_SGR_READY_TO USE is also unknown in my system.
So please delete the lines from .config:
ADD_NEW_REQUEST=0:TAG_SGR_REQ_READY_TO_USE-0
ADD_NEW_TOPIC=0:TAG_SGR_READY_TO_USE:W:1:0:sg_ready/ready_to_use

Please set
HISTORY_START_YEAR=2023
to avoid the message "History start year has been corrected"

The other messages can be ignored. Your system doesn't support the tags as mentioned (same with my system). And v3.32 also created the same messages...

Because I'm not using SG Ready... do the new topics help somehow?

Best regards
Thomas

jomifa, 2025-03-04:
Hello Thomas,
what I would like to see as a topic are values that show me whether SG Ready is switched on and which state it is.

[image]
[image]
[image]

s. At the moment only e3dc/sg_ready/used_power is displayed (see screenshots).

pvtom, 2025-03-06:
Hallo Joachim,
bei mir sieht das so aus (Ausgabe der Topics zu sg_ready, bei mir im Test mit "s10" beginnend konfiguriert):

[20250306065347.592163] s10/sg_ready/# s10/sg_ready/cooldown_end 0
[20250306065347.592664] s10/sg_ready/# s10/sg_ready/used_power   1300
[20250306065347.593267] s10/sg_ready/# s10/sg_ready/global_off   false
[20250306065347.639436] s10/sg_ready/# s10/sg_ready/1/provider/index 0
[20250306065347.639999] s10/sg_ready/# s10/sg_ready/2/provider/index 1
[20250306065347.640399] s10/sg_ready/# s10/sg_ready/3/provider/index 2
[20250306065347.642106] s10/sg_ready/# s10/sg_ready/4/provider/index 3
[20250306065347.642615] s10/sg_ready/# s10/sg_ready/1/provider/name  ModbusSgReadyHwProvider
[20250306065347.643535] s10/sg_ready/# s10/sg_ready/2/provider/name  SGReadyBuildIn
[20250306065347.643985] s10/sg_ready/# s10/sg_ready/3/provider/name  SGReadyOverEaton
[20250306065347.644420] s10/sg_ready/# s10/sg_ready/4/provider/name  SGReadyIOBox
[20250306065347.644600] s10/sg_ready/# s10/sg_ready/1/index          0
[20250306065347.644675] s10/sg_ready/# s10/sg_ready/2/index          1
[20250306065347.645146] s10/sg_ready/# s10/sg_ready/3/index          2
[20250306065347.645534] s10/sg_ready/# s10/sg_ready/4/index          3
[20250306065347.646270] s10/sg_ready/# s10/sg_ready/1/status         2
[20250306065347.646673] s10/sg_ready/# s10/sg_ready/2/status         2
[20250306065347.647634] s10/sg_ready/# s10/sg_ready/3/status         -1
[20250306065347.648106] s10/sg_ready/# s10/sg_ready/4/status         -1
[20250306065347.648590] s10/sg_ready/# s10/sg_ready/1/active         true
[20250306065347.648937] s10/sg_ready/# s10/sg_ready/2/active         false
[20250306065347.649018] s10/sg_ready/# s10/sg_ready/3/active         false
[20250306065347.649102] s10/sg_ready/# s10/sg_ready/4/active         true

# am Hauskraftwerk "an"
[20250306065451.994491] s10/sg_ready/# s10/sg_ready/2/active         true
# danach wieder auf "aus"
[20250306065459.234447] s10/sg_ready/# s10/sg_ready/2/active         false

D.h. es gibt eine Reaktion, wenn ich "an" / "aus" ändere in dem Menü, das Dein erstes gepostetes Bild zeigt.

Kannst Du das nachvollziehen?

Gruß Thomas

jomifa, 2025-03-06:
Hallo Thomas,
bei mir gibt es zurzeit keine Topics die mit e3dc/sg_ready beginnen. Wenn ich die Funktion am HKW ein- bzw. ausschalte, sehe ich keine e3dc/sg_ready Topics in meinem MQTT-Monitor.

***@***.***

Vielleciht stimmt irgendetwas nicht mit meiner .config. Ich hänge sie dir einmal an die Email an.

Grüße
Joachim

pvtom, 2025-03-06:
Hallo Joachim,
wenn Du das Programm neu startest, sollten sg_ready Daten kommen.
Passiert das?
Welchen Wert hast Du für INTERVAL konfiguriert.
Wenn der Wert z.B. 10 Sekunden beträgt und Du änderst innerhalb der 10 Sekunden die Einstellung am Hauskraftwerk an/aus, dann bekommt rscp2mqtt das gar nicht mit und sendet nichts.

Gruß Thomas

jomifa, 2025-03-06:
Hallo Thomas,
ist schon klar mit dem Interval Parameter. Mein Interval ist auf 30 Sekunden gesetzt und ich brauche mehr als 30 Sekunden, um vom HKW zum PC zu kommen. Bei einem Restart von rscp2mqtt bekomme ich nur den berichteten Wert für das Topic: „e3dc/sg_ready/used_power“ angezeigt.

Gibt es irgeneinde Definition, die im .config falsch sein könnte?

Gruß
Joachim

pvtom, 2025-03-07:
Hallo Joachim,
ich poste die Zeilen für .config hier nochmal. Oben habe ich gesehen, dass einige Unterstriche nicht dargestellt werden, evtl. liegt es daran.

```
ADD_NEW_REQUEST_AT_START=0:TAG_SGR_REQ_HW_PROVIDER_LIST-0
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_INDEX:_:1:0:sg_ready/1/provider/index
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_NAME:_:1:0:sg_ready/1/provider/name
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_INDEX:_:1:0:sg_ready/2/provider/index
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_NAME:_:1:0:sg_ready/2/provider/name
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_INDEX:_:1:0:sg_ready/3/provider/index
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_NAME:_:1:0:sg_ready/3/provider/name
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_INDEX:_:1:0:sg_ready/4/provider/index
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_NAME:_:1:0:sg_ready/4/provider/name
ADD_NEW_REQUEST=0:TAG_SGR_REQ_COOLDOWN_END-0
ADD_NEW_TOPIC=0:TAG_SGR_COOLDOWN_END:_:1:0:sg_ready/cooldown_end
ADD_NEW_REQUEST=0:TAG_SGR_REQ_USED_POWER-0
ADD_NEW_TOPIC=0:TAG_SGR_USED_POWER:W:1:0:sg_ready/used_power
ADD_NEW_REQUEST=0:TAG_SGR_REQ_GLOBAL_OFF-0
ADD_NEW_TOPIC=0:TAG_SGR_GLOBAL_OFF:_:1:0:sg_ready/global_off
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_INDEX:0-0
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_REQ_STATE-1
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_INDEX:_:1:0:sg_ready/1/index
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_STATE:_:1:0:sg_ready/1/status
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_AKTIV:_:1:0:sg_ready/1/active
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_INDEX:1-2
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_REQ_STATE-3
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_INDEX:_:1:0:sg_ready/2/index
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_STATE:_:1:0:sg_ready/2/status
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_AKTIV:_:1:0:sg_ready/2/active
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_INDEX:2-4
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_REQ_STATE-5
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_INDEX:_:1:0:sg_ready/3/index
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_STATE:_:1:0:sg_ready/3/status
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_AKTIV:_:1:0:sg_ready/3/active
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_INDEX:3-6
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_REQ_STATE-7
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_INDEX:_:1:0:sg_ready/4/index
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_STATE:_:1:0:sg_ready/4/status
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_AKTIV:_:1:0:sg_ready/4/active
```
Gruß Thomas

jomifa, 2025-03-07:
Hallo Thomas,
das war die Lösung. 😉

Jetzt müsste ich nur noch wissen was sich hinter den Topics verbirgt.

Ich habe noch eine andere Frage:

Ich bekomme zurzeit in MQTT das Topic „e3dc/ema/emergency_power_available“ nicht ausgegeben, obwohl ich es in der .config erzwungen habe. Hast du eine Idee dazu, woran es liegen könnte:

#-Forced Topics
#--will be published with every cycle
FORCE_PUB=e3dc/[a-z]+/power
FORCE_PUB=e3dc/battery/soc
FORCE_PUB=e3dc/pvi/.*/string_[1,2]
FORCE_PUB=e3dc/system/derate/.*
FORCE_PUB=e3dc/grid_in_limit.*
FORCE_PUB=e3dc/battery/cycles
#FORCE PUB=e3dc/rscp2mqtt/version
FORCE_PUB=e3dc/ems/emergency_power_available
FORCE_PUB=e3dc/ems/emergency_power_status
#FORCE_PUB=e3dc/ems/weather_regulation
FORCE_PUB=e3dc/wallbox/sun_mode
FORCE_PUB=e3dc/wallbox/plugged

Grüße
Joachim

lorbetzki, 2025-03-07:
Hallo,

```
ADD_NEW_REQUEST_AT_START=0:TAG_SGR_REQ_HW_PROVIDER_LIST-0
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_INDEX:_:1:0:sg_ready/1/provider/index
ADD_NEW_TOPIC=TAG_SGR_HW_PROVIDER:TAG_SGR_NAME:_:1:0:sg_ready/1/provider/name
ADD_NEW_REQUEST=0:TAG_SGR_REQ_COOLDOWN_END-0
ADD_NEW_TOPIC=0:TAG_SGR_COOLDOWN_END:_:1:0:sg_ready/cooldown_end
ADD_NEW_REQUEST=0:TAG_SGR_REQ_USED_POWER-0
ADD_NEW_TOPIC=0:TAG_SGR_USED_POWER:W:1:0:sg_ready/used_power
ADD_NEW_REQUEST=0:TAG_SGR_REQ_GLOBAL_OFF-0
ADD_NEW_TOPIC=0:TAG_SGR_GLOBAL_OFF:_:1:0:sg_ready/global_off
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_INDEX:0-0
ADD_NEW_REQUEST=TAG_SGR_REQ_DATA:TAG_SGR_REQ_STATE-1
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_INDEX:_:1:0:sg_ready/1/index
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_STATE:_:1:0:sg_ready/1/status
ADD_NEW_TOPIC=TAG_SGR_DATA:TAG_SGR_AKTIV:_:1:0:sg_ready/1/active
```

das reicht bei mir aus um den SG Ready Status anzeigen zu lassen. 
sg_ready/1 ist der Status laut Anlage, unabhängig vom verwendeten SG-Ready System
sg_ready/2 ist der Status vom Built-IN (also dem Relais Modul)
sg_ready/3 ist der Status von einem Eaton Modul (also Funk)
sg_ready/4 ist der Status von einer IO/Box.

Vielen Dank für die coole Funktion!

closed 2025-03-11 by pvtom