pvtom/rscp2mqtt #99: consumption of the wallbox in sunshine mode

jomifa, 2025-03-02:
Hello,
the consumption of the wallbox in sunshine mode is not displayed correctly. The values between RSCP2MQTT and the E3DC portal do not match.

RSCP2MQTT Version: 3.32

[image]
[image]

pvtom, 2025-03-03:
Hello,
when was rscp2mqtt started? On the day of charging or during charging? Background: rscp2mqtt calculates the values ‘last_charging’ and ‘day’ as the difference between the values ‘e3dc/wallbox/energy/solar’ before and after/during charging. The home power station can also deliver the values with a delay. Unfortunately, I cannot reproduce this as I do not have an E3DC wallbox.
Best regards
Thomas

jomifa, 2025-03-03:
Hello Thomas,
In this case, rscp2mqtt was restarted during the charging process.

Regards
Joachim

pvtom, 2025-03-03:
Hello Joachim,
okay, that's the reason for the wrong values.
To ensure that the values survive a restart, set RETAIN_FOR_SETUP=true in .config.
Best regards
Thomas

jomifa, 2025-03-03:
Hello Thomas,
this parameter is already included in the .config

Regards
Joachim