pvtom/rscp2mqtt #100: Berechnung Reserve %

MyGitIT, 2025-03-03:
Hallo Thomas,

kurze Frage, in der Berechnung der Reserve mit % musst du ja irgendwo die tatsächliche Kapazität der Batterie haben, steht die auch in einem Topic oder mit welchen Topics berechnest du die?

Gruß Sven

pvtom, 2025-03-03:
Hallo Sven,
meinst Du die Batterie-Reserve, die man mit "e3dc/set/reserve/percent" setzen kann?
Dieser Prozentwert wird dem Hauskraftwerk übergeben und das "weiß" dann, was es zu tun hat.
Die tatsächliche Batteriekapazität kann man mit unterschiedlichen Topics, die "capacity" im Namen haben, ermitteln, z.B. "e3dc/battery/usable_capacity" in Ah.
Gruß
Thomas

MyGitIT, 2025-03-03:
Hi Thomas,

ahh, ich dachte du würdest da rechnen, ich brauche nämlich die noch tatsächliche Kapazität des Akkus. Dann muss ich mir mal die richtigen Topics zum Rechnen suchen.

Gruß Sven

closed 2025-03-11 by pvtom