pvtom/rscp2mqtt #101: Docker: Fehler Not supported by the device: Container >TAG_PVI_DATA<

LordMM74, 2025-03-04:
Hallo,
ich habe mich die letzten Tage mit dem Ding hier beschäftigt.

Ich hab unter Docker 
- InfluxDB
- mosquitto
- Node-Red
- Grafana
installiert und konfiguriert und dann habe ich versicht das rscp2mqtt ebenfalls als Container zum laufen zu kriegen.

Die .config ist korrekt gemountet und wird augenscheinlich auch erkannt.

Leider bin ich an der Stelle dann mit meinem Latein am Ende.
Es landen auch Daten in der InfluxDB ... aber nur 7 measurements. Die die per MQTT über den Broker gehen müssten kommen nicht an.
Weder in mosquitto noch im Logfile von Node-Red seh ich was.

Kann mir da wer helfen?
Danke

```
rscp2mqtt [3.34.influxdb]
E3DC system >xxxxxxxxxxxxxxxx:5033< user: >xxxxxxxxxxxxxxxx<
MQTT broker >xxxxxxxxxxxxxxxx:1883< qos = >0< retain = >✗< tls >✗< client id >e3dc< prefix >e3dc<
INFLUXDB v2 >xxxxxxxxxxxxxxxx:8086< orga = >XXXX< bucket = >e3dc< measurements = >e3dc< and >e3dc_meta<
Requesting PVI ✓ | PM (0) | DCB ✓ (1 battery string) | Wallbox ✗ | Interval 5 | Autorefresh ✓ | Raw data ✗ | Logging ON
Stdout to pipe/file
[2025-03-04 22:04:24] pid=1 ppid=0 RscpMqttMain.cpp(2368) Not supported by the device: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_COS_PHI< received >NOT_AVAILABLE< [1]
[2025-03-04 22:04:24] pid=1 ppid=0 RscpMqttMain.cpp(2368) Not supported by the device: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_VOLTAGE_MONITORING< received >NOT_AVAILABLE< [1]
[2025-03-04 22:04:24] pid=1 ppid=0 RscpMqttMain.cpp(2368) Not supported by the device: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_FREQUENCY_UNDER_OVER< received >NOT_AVAILABLE< [1]
```

pvtom, 2025-03-06:
Hallo,
die Ausgaben "Not supported..." sind keine Fehler, zeigen aber, dass der Kontakt von rscp2mqtt zum Hauskraftwerk funktioniert. Weitere Ausgaben bekommst Du, wenn Du rscp2mqtt -v ausführst.
Da es keine weiteren Fehlermeldungen gibt, gehe ich davon aus, dass der Kontakt zum Broker auch da ist. Keine Ahnung, was da schief geht. Kannst Du den Broker von außen erreichen und per mosquitto_pub Daten übergeben und die beim Subscribe sehen?
Wie ist denn die Konstellation genau? Den Weg InfluxDB->Grafana kenne ich, um die Werte darstellen/auswerten zu können. Was soll Node-Red machen?
Welche Werte kommen denn in der Influx an?
Gruß Thomas

LordMM74, 2025-03-06:
Ich hab mich im ersten Schritt an das hier gehalten

https://github.com/MrIceman11/e3dc-to-grafana

pvtom, 2025-03-06:
Okay, da wird eine Lösung von MQTT über Node-Red zur InfluxDB beschrieben. Na ja, rscp2mqtt kann ja direkt in die InfluxDB schreiben. Die Lösung ist also einfacher (ohne Node-Red) möglich.

rscp2mqtt versorgt parallel den MQTT Broker und die InfluxDB.

DISABLE_MQTT_PUBLISH=true bewirkt, dass nichts an den MQTT Broker gepublisht wird. Dann würde nur die InfluxDB beschrieben.

Gruß Thomas

LordMM74, 2025-03-06:
Danke. Das macht es einfacher nicht den Umweg über einen Broker zu gehen.
Ich behalte mir das mit dem Broker aber noch im Hinterkopf weil ich noch OpenHab laufen habe weil dort meine anderen Geräte (Sensoren, Mähroboter, Poolsteuerung, 3D Drucker usw) auflaufen.

Gibt’s für das -v in der .config eine Variable?
-v kann ich ja nicht als Parameter im Docker run verwenden weil -v dort für das Mounten von Volumes benutzt wird (in Docker bin ich noch neu).

pvtom, 2025-03-07:
Leider kann man -v nicht per .config setzen.
Möglich wäre sich interaktiv in den Container einzuklinken und dann rscp2mqtt -v auszuführen.
Hast schon probiert mit dem MQTT Broker direkt zu kommunizieren? Z.B. mit den Mosquitto-Clients mosquitto_pub / mosquitto_sub.
Gruß Thomas

LordMM74, 2025-03-07:
Nein. 
Aktuell bin ich mit DISABLE_MQTT_PUBLISH=true unterwegs damit die Daten ohne Umwege in die InfluxDB geschrieben werden.

Ich hoffe mal das die Daten die in die DB geschrieben werden die gleichen sind die auch an den Broker gehen würden. 
Aktuell kämpfe ich noch mit den Neuerungen InfluxDB v1 zu v2 … und damit verbunden Flux als neue Query Language.
SQL mach ich ja eigentlich schon länger … aber leider ist Flux da doch anders gestrickt und Grafana hat für Flux leider keinen wirklichen Editor.

pvtom, 2025-03-08:
Tipp: INFLUXDB_TOPIC in der .config schränkt die Menge der Topics, die an die InfuxDB gesendet werden, ein.
Wenn Du alle Topics haben möchtest, alle (können mehrere sein) INFLUXDB_TOPIC - Einträge entfernen.

Gruß Thomas

LordMM74, 2025-03-08:
Damit hab ich schon getestet … aktuell kommen aber gar keine Daten in der Datenbank an. 
Deaktiviere ich die Datenbank und gebe die Daten an den Broker kommen dort Daten an.
Ich hab die Config für die Datenbank schon zigfach geprüft.

Ich hab mehrfach 1:1 die gleiche Config in Grafana als Datasource eingetragen … Grafana kommt auf die Daten.
Das Schreiben der Daten aus rscp geht mit den identischen Daten (Token, Organisation, Adresse, Bucket) geht nicht.

LordMM74, 2025-03-08:
Mein .config schaut so aus:

```
#--dry run
DISABLE_MQTT_PUBLISH=true

ENABLE_INFLUXDB=true
INFLUXDB_HOST=x.x.x.x
INFLUXDB_PORT=8086
INFLUXDB_VERSION=2
INFLUXDB_MEASUREMENT=e3dc
INFLUXDB_MEASUREMENT_META=e3dc_meta
INFLUXDB_TOPIC=e3dc/[a-z]+/power
INFLUXDB_TOPIC=e3dc/day/.*
INFLUXDB_TOPIC=e3dc/week/.*

INFLUXDB_2_ORGA=XXX
INFLUXDB_2_BUCKET=XXX
INFLUXDB_2_TOKEN=XXX
```

Meine Werte hab ich mit X geschwärzt.

Danach starte ich das Ding mit -v

```
rscp2mqtt [3.34.influxdb]
E3DC system >192.168.178.69:5033< user: >xxxxxxxxxx<
History year range from 2022 to 2025
MQTT broker >192.168.178.111:1883< qos = >0< retain = >✗< tls >✗< client id >e3dc< prefix >e3dc<
INFLUXDB v2 >http://192.168.178.111:8086< orga = >XXXX< bucket = >XXXX< measurements = >e3dc< and >e3dc_meta<
Requesting PVI ✓ | PM (0) | DCB ✓ (1 battery string) | Wallbox ✗ | Interval 5 | Autorefresh ✓ | Raw data ✗ | Logging ON
DISABLE_MQTT_PUBLISH / DRYRUN mode
[2025-03-08 16:22:12] pid=17 ppid=8 RscpMqttMain.cpp(3525) Connecting to server 192.168.178.69:5033
[2025-03-08 16:22:12] pid=17 ppid=8 RscpMqttMain.cpp(3532) Success: E3DC connected.
[2025-03-08 16:22:12] pid=17 ppid=8 RscpMqttMain.cpp(2252) RSCP authentication level 10
[2025-03-08 16:22:12] pid=17 ppid=8 RscpMqttMain.cpp(2812) Connecting to broker 192.168.178.111:1883
[2025-03-08 16:22:12] pid=17 ppid=8 RscpMqttMain.cpp(2835) Success: MQTT broker connected.
[2025-03-08 16:22:12] pid=17 ppid=8 RscpMqttMain.cpp(462) MQTT: starting listener loop
[2025-03-08 16:22:14] pid=17 ppid=8 RscpMqttMain.cpp(2368) Not supported by the device: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_COS_PHI< received >NOT_AVAILABLE< [1]
[2025-03-08 16:22:14] pid=17 ppid=8 RscpMqttMain.cpp(2368) Not supported by the device: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_VOLTAGE_MONITORING< received >NOT_AVAILABLE< [1]
[2025-03-08 16:22:14] pid=17 ppid=8 RscpMqttMain.cpp(2368) Not supported by the device: Container >TAG_PVI_DATA< Tag >TAG_PVI_REQ_FREQUENCY_UNDER_OVER< received >NOT_AVAILABLE< [1]
[2025-03-08 16:22:14] pid=17 ppid=8 RscpMqttMain.cpp(538) Publish topic >**e3dc/solar/power**< payload >4562< unit >W< mqtt:✗ influx:✓
[2025-03-08 16:22:14] pid=17 ppid=8 RscpMqttMain.cpp(538) Publish topic >e3dc/battery/power< payload >0< unit >W< mqtt:✗ influx:✓
[2025-03-08 16:22:14] pid=17 ppid=8 RscpMqttMain.cpp(538) Publish topic >e3dc/home/power< payload >472< unit >W< mqtt:✗ influx:✓
[2025-03-08 16:22:14] pid=17 ppid=8 RscpMqttMain.cpp(538) Publish topic >e3dc/grid/power< payload >-4090< unit >W< mqtt:✗ influx:✓
```

Das Ding läuft dann weiter in die Werte kommen auch wenn er mit TAG_PVI_DATA nix anfangen kann.

Geht ich auf die InfluxDB und schau mal z. B. nach **e3dc/solar/power**

```
from(bucket: "e3dc")
  |> range(start: v.timeRangeStart, stop: v.timeRangeStop)
  |> filter(fn: (r) => r["_measurement"] == "e3dc/solar/power")
  |> filter(fn: (r) => r["_field"] == "value")
  |> aggregateWindow(every: v.windowPeriod, fn: mean, createEmpty: false)
  |> yield(name: "mean")
```

dann ist da nix.
Am Broker kommen die Daten (wenn ich DISABLE_MQTT_PUBLISH=false) setze an und schreibe ich die dann z. B. über Node-Red aus in die DB kommen sie dort auch an.
Direkt aus RSCP kommt nichts in der Datenbank an auch mit -v kommen allerdings keinerlei weitere Fehler aus das TAG_PVI_DATA Zeug.

LordMM74, 2025-03-08:
Danke für die Hilfe.
Inzwischen habe ich Daten sowohl im Broker als auch in der DB … der Fehler aus dem ersten Kommentar ist aber nach wie vor vorhanden.

pvtom, 2025-03-08:
Prima. Freut mich, dass es jetzt läuft. Wie konntest Du das Problem beheben?
Die Fehler mit "Not supported..." sind nur Hinweismeldungen, weil das Hauskraftwerk die Tags nicht unterstützt. Ist bei mir auch so...
Gruß Thomas

LordMM74, 2025-03-08:
Ich hatte mich beim Start nach dieser Anleitung durchgehangelt 

https://github.com/MrIceman11/e3dc-to-grafana

Der verweist ja auch auf dein Projekt.
Er nutzt aber MQTT und Node-Red um die Daten nach Influx zu bringen.

Als ich dann bei rscp war nach der Anleitung hab ich beim dann gelesen das der Umweg über Node-Red ja eigentlich unnötig ist … Ich hab dann viel rumprobiert.
Letztendlich nutzt  die andere Anleitung Node-Red und ich weiß nicht genau warum und wie … aber letztendlich gabs kamen die Daten nicht an.

Ich hab schlicht alle Container gestoppt … hab die InfluxDB per CLI komplett leer gemacht.
Node-Red hab ich dann komplett gelöscht und dann alles wieder hochgefahren und dann liegen plötzlich die Daten aus rscp in der Datenbank auf.

Ich vermute das das bucket in Influx irgendwie mit den verschiedenen Dingen nicht zurechtkam … warum auch immer. Ich hab das Bucket selber nur geleert … nicht gelöscht. Alle Tokens hab ich gelassen (an denen lags also nicht).

closed 2025-03-11 by pvtom