pvtom/rscp2mqtt #103: Cell Information

genezig, 2025-03-14:
Hi there,

using another tool (RSCPGUI) I discovered that it is possible to get information about the batteries on cell level (voltage, temperature) via RSCP. Is it possible to add this functionality to rscp2mqtt?

Thanks for the great work!

Best regards

pvtom, 2025-03-14:
Hi,
no problem. This functionality is still available.
Maybe it's disabled by
DCB_REQUESTS=false
in the .config file.
Please delete the line or set the value to true and you will get topics: e3dc/battery/dcb/#

Best regards
Thomas

genezig, 2025-03-15:
Thank you for the quick reply. I have DCB_REQUESTS=true in my .config file and I get information about the batteries (i.e. soh). But in RSCPGUI there is also information on cell level, i.e. voltage of 3.238 V (every battery seems to consist of 16 cells). Also, I get two temperatures per battery while with rscp2mqtt I only get the max. and the min. temperature of the whole battery system. There are tags named BAT_DCB_ALL_CELL_VOLTAGES and BAT_DCB_ALL_CELL_TEMPERATURES which seem to carry the information I am looking for.

Best regards

pvtom, 2025-03-23:
it will be available with release v3.36

pvtom, 2025-03-29:
v3.36 is available