pvtom/rscp2mqtt #105: mosquitto_pub -> influxdb (./request_days.sh)

LordMM74, 2025-03-20:
Das script ./request_days.sh ermittelt die Daten für einen übergebenen Zeitraum und per mosquitto_pub werden diese allem Anschein nach an den Broker veröffentlicht.

Gibt mosquitto_pub die Daten auch an die InfluxDB wenn diese per .config als Ausgabe konfiguriert ist?

Wenn nein ... gibts eine Möglichkeit historische Daten gezielt ermitteln zu lassen und diese dann in die InfluxDB zu kriegen oder muss ich dann Umweg über den Broker gehen ... was letztendlich umständlich wäre weil man dann dort erstmal wieder aufpassen muss das man die topics alle richtig hat.

pvtom, 2025-03-20:
Hallo,
ja, die Tagesdaten werden auch in die InfluxDB geschrieben.
Dabei wird der Timestamp auf das Tagesdatum und 0:00 Uhr gesetzt.
Es kommt dann darauf an, wie die InfluxDB konfiguriert ist, ob die Daten gehalten oder gleich wieder gelöscht werden. Stichwort Retention Policy.
Gruß Thomas

LordMM74, 2025-03-20:
Probier ich aus.
Die Retention Policy sollte erst mal kein Problem sein. Hab ich erst mal auf nie löschen gesetzt.

MyGitIT, 2025-03-24:
Guten Morgen, anscheinend habe ich ein ähnliches Problem, seit dem 19.03 werden weder History noch Systemdaten in die Influxdb geschrieben. Erst wenn ich die Topics mit Force eintrage werden sie in die DB geschrieben.

LordMM74, 2025-03-24:
Ich hab das heute mal getestet.

Bei mir läuft rscp2mqtt ja in einem Container und Mosquitto in einem eigenen Container.
Das script request_days.sh kann daher ja nur im Mosquitto Container laufen (weil mosquitto_pub nur dort funktioniert).

Ich krieg da aber aktuell noch einen Fehler

`
date: invalid date '2025-01-02 + 1 day'
request_days.sh: line 37: can't open : no such file
`

When ich einfach mal mit dem MQTT Explorer einen publish auf e3dc/set/request/day/2025-01-01 mach dann scheint das grds. zu funktionieren:

[image]

Allerdings kommen die Daten nicht in der influxdb an sondern nur im Broker.

LordMM74, 2025-03-24:
Ich hab das heute mal durchgespielt.

Bei mir läuft rscp2mqtt in einem Container und Mosquitto in einem eigenen Container (d. h. mosquitto_pub funktioniert auch nur im Mosquitto-Container).

Bsp.:
`
mosquitto_pub -h xxx.xxx.xxx.xxx -p 1883 -t "e3dc/set/request/day" -m "2025-01-01"
`
(das entspricht ja dem was das Script in Schleife tut)

Das funktioniert bei mir wie erwartet und die Daten wandern auch direkt in die InfluxDB (sind aber auch im MQTT Explorer zu sehen).
Das Script selbst ausgeführt im Mosquitto-Container führt allerdings zu einem Fehler in Zeile 37 beim Hochzählen des Datums.

LordMM74, 2025-03-24:
Ich hab mir jetzt schnell mal ne Linux VM auf meinem Unraid (Docker) hochgezogen und dort rscp2mqtt und den Mosquitto MQTT Client installiert.
Dort läuft das Script astrein.

closed 2025-03-24 by LordMM74

pvtom, 2025-03-25:
> Guten Morgen, anscheinend habe ich ein ähnliches Problem, seit dem 19.03 werden weder History noch Systemdaten in die Influxdb geschrieben. Erst wenn ich die Topics mit Force eintrage werden sie in die DB geschrieben.

Ob Daten an die InfluxDB gesendet werden, hängt von den Parametern INFLUXDB_TOPIC und FORCE_PUB ab.
Wenn kein INFLUXDB_TOPIC - Eintrag in der .config existiert, werden alle Topics in die InfluxDB geschrieben und zwar die, die sich seit dem letzten Schreiben vom Wert her verändert haben. Mit FORCE_PUB kann man Topics festlegen, die in jedem Zyklus rausgeschriebn werden sollen, d.h. auch wenn der Wert sich nicht geändert hat.
Wenn man einen oder mehrere INFLUXDB_TOPIC - Einträge definiert hat, gelten diese. Es werden dann nur noch die definierten Topics geschrieben.
(Meine Testkonstellation: rscp2mqtt 3.35.influxdb, InfluxDB V1)
Gruß Thomas