pvtom/rscp2mqtt #108: Topic e3dc/wallbox/suspended will not set correctly

jomifa, 2025-04-26:
Hello,
I set the topic: “e3dc/set/wallbox/suspended” to true with the assumption that the topic: “e3dc/wallbox/suspended” will also be set to true after the next interval. This is not the case. In the MQTT Explorer you can see that “e3dc/wallbox/suspended” is sometimes not set at all or is set incorrectly. 

RSCP2MQTT Release 3.35 is running.

[image]

[image]

pvtom, 2025-04-27:
Hello Joachim,
unfortunately I can't test this. I don't have an E3/DC wallbox.
Is there a reaction from the wallbox? e3dc/wallbox/suspended is supposed to stop the charging process. Does that happen?
Alternatively, e3dc/wallbox/toggle with value 1 should switch between the state of charging (on to off, off to on). Have you already tried this?
Best regards
Thomas

jomifa, 2025-04-27:
Hello Thomas,
yes, the suspension has already worked. However, I am not sure whether the topic “e3dc/wallbox/sundended” has also been set to true. Unfortunately, the behavior is not conclusive. I do not know the topic “e3dc/wallbox/toggle”. It is also not in your table. Is my expectation when using “e3dc/set/wallbox/suspended” to “ e3dc/wallbox/suspended” correct?

Regards
Joachim

pvtom, 2025-04-27:
Hello Joachim,
yes, “e3dc/set/wallbox/suspended” should change the value for “e3dc/wallbox/suspended" as the reaction of the wallbox. Maybe some seconds (request cycles) later... But I will check the logic again.

I have corrected my last comment: "set" was missing, sorry.
You can toggle the charging state with "e3dc/set/wallbox/toggle", which also should be an effect to "e3dc/wallbox/suspended".

One conclusion, however, is that the wallbox can be controlled by rscp2mqtt :-)

Best regards
Thomas

pvtom, 2025-04-27:
Logic is...
1. e3dc/wallbox/suspended has the value true and user sends e3dc/set/wallbox/suspended true -> no change because e3dc/wallbox/suspended is already true
2. e3dc/wallbox/suspended has the value false and user sends e3dc/set/wallbox/suspended true -> rscp2mqtt sends "toggle the state, here to start charging"
3. e3dc/wallbox/suspended has the value true and user sends e3dc/set/wallbox/suspended false > rscp2mqtt sends "toggle the state, here to stop charging"
4. e3dc/wallbox/suspended has the value false and user sends e3dc/set/wallbox/suspended false -> no change because e3dc/wallbox/suspended is already false

Problem: If e3dc/set/wallbox/suspended will not be changed by the wallbox (by an error or something else), the cases 1 to 4 will not work correctly!

e3dc/set/wallbox/toggle will change the state by the wallbox itself, it works without checking/knowing the state of e3dc/wallbox/suspended by rscp2mqtt ... maybe the better way...

I hope this helps a bit to understand the behaviour.

jomifa, 2025-04-27:
Hello Thomas,
it works reliably with the toggle. Thank you for your support ;-)

Greetings
Joachim

pvtom, 2025-04-28:
Hello Joachim,
that's a good thing!
Does the value of e3dc/wallbox/suspended change in response to the toggle command?
Best regards
Thomas

jomifa, 2025-04-28:
Hello Thomas,
yes, e3dc/wallbox/suspended works correctly in conjunction with e3dc/set/wallbox/toggle.

Regards
Joachim

closed 2025-07-07 by pvtom