pvtom/rscp2mqtt #110: Daily Energy discrepancy

Xenos1104, 2025-06-18:
Hello everyone,

first of all thank you for developing this. Really useful, since E3DC doesn't open any other possibilities besides Modbus, which is really limited.

I am having an issue regarding the daily energy meters especially for "daily home consumption" and "daily battery discharge". Those two values are - in contrast to the other daily energy meters - not correlated to te E3DC App.

For example I am listing my values of today as follows:

MQTT-Topic | Value from rscp2mqtt | Value in E3DC App

e3dc/solar/energy                     |   75.78   |   75.78
**e3dc/home/energy                    |   10.98  |   11.85**
e3dc/grid/energy/in                  |    60.42  |    60.43
e3dc/grid/energy/out                |      0.08  |      0.09
e3dc/battery/energy/charge      |     5.81  |      5.81
**e3dc/battery/energy/discharge |     3.02  |      4.03**

As you can see the values for "solar/energy" or "grid/energy/in" or other values differ from the app by 0.01 max which I am okay with since there is rounding and time of reading may be off by some minutes etc.

The values in e3dc/home/energy and e3dc/battery/energy/discharge are off by nearly 1 kWh over a day and even those differences are not lining up.

Does anyone know something about this?

I would be glad to hear an explanation about this. Maybe I am reading the wrong tags or something... 

Have a nice day and thanks in advance!

pvtom, 2025-06-18:
Hello,
I have compared my values from today. They match exactly, there are no deviations, not even in the decimal places.
I cannot explain the reason for the deviation in your case. What values does the home power station show on the display?
Best regards
Thomas

pvtom, 2025-06-18:
Could you please check the time on your computer where rscp2mqtt is running. Do you use the docker version? (set TZ variable)

closed 2025-07-07 by pvtom