pvtom/rscp2mqtt #111: topic "e3dc/set/wallbox/number_phases"

CYBERDIVER76, 2025-07-13:
Hello,
i have a question about the topic “e3dc/set/wallbox/number_phases”. I can currently only send the value 1 or 3 via my SmartHome (MQTT). However, I can also set the WB to “Auto” in the menu of the E3/DC (or the app). Is this also somehow possible via MQTT? With one phase, the minimum power is 1.3kW and with 3 phases it is 4.2kW. I would like to charge via the SmartHome up to a certain battery level on “one” phase (if there is enough sun) and then switch to “Auto”, so that the system can see for itself how much surplus is available for the car.

Kind regards
Sebastian

CYBERDIVER76, 2025-08-26:
Hello Thomas,

ADD_NEW_SET_TOPIC=TAG_WB_REQ_DATA:TAG_WB_REQ_SET_AUTO_PHASE_SWITCH_ENABLED:0:set/wallbox/auto_phase_switch:^true|1$:true:^false|0$:false#Bool
Added to .config. Sent from Smarthome to E3DC OK.

Unfortunately, receiving the values (E3DC->Smarthome) isn't working. I also don't see any messages in MQTT.fx from E3DC. Even though I added your following lines.

ADD_NEW_REQUEST=TAG_WB_REQ_DATA:TAG_WB_INDEX:0-0
ADD_NEW_REQUEST=TAG_WB_REQ_DATA:TAG_WB_REQ_AUTO_PHASE_SWITCH_ENABLED-1
ADD_NEW_TOPIC=TAG_WB_REQ_DATA: TAG_WB_AUTO_PHASE_SWITCH_ENABLED:_:1:0:wallbox/auto_phase_switch

I would like to help/continue here. Is there a log file that I can view or send?

Or is there another user here who could possibly test this again?

Kind regards
Sebastian

CYBERDIVER76, 2025-08-26:
Hello Thomas,

sorry, but it still doesn't work.

```
ADD_NEW_REQUEST=TAG_WB_REQ_DATA:TAG_WB_INDEX:0-0
ADD_NEW_REQUEST=TAG_WB_REQ_DATA:TAG_WB_REQ_AUTO_PHASE_SWITCH_ENABLED-1
ADD_NEW_TOPIC=TAG_WB_DATA:TAG_WB_AUTO_PHASE_SWITCH_ENABLED:_:1:0:wallbox/auto_phase_switch
```
Best regards
Sebastian

pvtom, 2025-08-26:
Hmm, okay, let's try:
ADD_NEW_TOPIC=TAG_WB_DATA:TAG_WB_AUTO_PHASE_SWITCH_ENABLED:0:_:1:0:wallbox/auto_phase_switch
(with index 0)

Another step is to set 
RAW_DATA=true
in .config, so you will get e3dc/raw/TAG_* output. 
Maybe you will find a topic with TAG_WB_AUTO_PHASE_SWITCH_ENABLED.

Best regards
Thomas

CYBERDIVER76, 2025-08-27:
Hi Thomas,

Now it works both ways!

Here are the additional parameters/lines in my configuration file (for those who want to replicate it)

```
# auto_phase_switch | smarthome (MQTT) -> E3DC
ADD_NEW_SET_TOPIC=TAG_WB_REQ_DATA:TAG_WB_REQ_SET_AUTO_PHASE_SWITCH_ENABLED:0:set/wallbox/auto_phase_switch:^true|1$:true:^false|0$:false#Bool

# auto_phase_switch | E3DC -> smarthome (MQTT)
ADD_NEW_REQUEST=TAG_WB_REQ_DATA:TAG_WB_INDEX:0-0
ADD_NEW_REQUEST=TAG_WB_REQ_DATA:TAG_WB_REQ_AUTO_PHASE_SWITCH_ENABLED-1
ADD_NEW_TOPIC=TAG_WB_DATA:TAG_WB_AUTO_PHASE_SWITCH_ENABLED:1:_:1:0:wallbox/auto_phase_switch
```

Best regards
Sebastian